OpenAstroTech/OpenAstroTracker #2: Shopping List

HenriqueOz, 2020-02-24:
I did some googling to find parts here in Brazil and I think it can help other Brazilians. Maybe you could include a column for each country so people can discover if this project is feasible easier and we can share cheaper prices! 

[New Shopping list](https://docs.google.com/spreadsheets/d/1OqxZIwVMUGhERcs3vZSwPZy3x14aWcIlta8j2wWsevc/edit?usp=sharing)

OpenAstroTech, 2020-02-25:
Thats a really good idea, i'll add it

closed 2020-02-25 by OpenAstroTech

OpenAstroTech, 2020-02-25:
I changed the step converter to [this one](https://produto.mercadolivre.com.br/MLB-756355702-regulador-de-tenso-ajustavel-lm2596-dc-dc-display-step-down-_JM?quantity=1#reco_item_pos=3) The screen makes it much easier to set the voltage when you dont have a voltmeter