OpenAstroTech/OpenAstroTracker #5: Arduino Code

ztwillis, 2020-03-05:
In the Calculations file in the Arduino code the pins were different than those in Init:
I believe this should be what it reads:

if (DECheat == 1) {
  stepperDEC.enableOutputs();
  digitalWrite(15, HIGH);
  digitalWrite(16, HIGH);
  digitalWrite(17, HIGH);
  digitalWrite(18, HIGH);
 
Also when I enter my HA from based on Polaris in in stellarium the declination axis(initial position starting perpendicular to RA axis) goes nearly 180 degrees when I enter my HA and go to the Polaris tab and select it.  Thanks in advance!

OpenAstroTech, 2020-03-05:
> different than those in Init

oh right thanks, i changed it

> goes nearly 180 degrees when I enter my HA and go to the Polaris tab and select it

Thats weird, that cant happen unless you entered some coordinates before using the POL menu

ztwillis, 2020-03-05:
Honestly it's probably user error, mostly I just wanted to let you know about the Arduino code, thanks for the quick response and all of the work you're doing!

commit 62008469 (2020-04-15): Fixes #5

I am an idiot, and know beans about mutexes.
closed 2020-03-05 by ztwillis