OpenAstroTech/OpenAstroTracker #14: Button

tomv79, 2020-03-11:
The left button does not switch throught the menues.
It has the same effect as the down button.
Sometimes the left button switches the menue.

OpenAstroTech, 2020-03-11:
Go to the second tab of the code "a_inits" and find line 36 "if (adc_key_in < 600)  return btnLEFT;"

Try if changing the value in that line to 700 or 800 fixes it

tomv79, 2020-03-11:
Yes the left button now works.
But the select button seem to be the left now.

OpenAstroTech, 2020-03-11:
Try increasing its value in the code till it works

tomv79, 2020-03-12:
I have make following changes that works for me well!!

if (adc_key_in > 1000) return btnNONE;
  if (adc_key_in < 20)   return btnRIGHT;
  if (adc_key_in < 150)  return btnUP;
  if (adc_key_in < 300)  return btnDOWN;
  if (adc_key_in < 500)  return btnLEFT;
  if (adc_key_in < 700)  return btnSELECT;

now the keys works fine!

OpenAstroTech, 2020-03-13:
Great! I'll mention this in the upcoming troubleshooting guide, i think some of the LCDs need different values.

Gonna close this

closed 2020-03-13 by OpenAstroTech