OpenAstroTech/OpenAstroTracker #46: OATcontrol crashing on startup

BenJuan26, 2020-08-03:
I've downloaded the latest release and extracted OATcontrol. When I try to run `OATControl.exe`, it crashes instantly; no error message, no error report, etc. I attached Visual Studio to the `.exe` and got a bit more information. It seems to be a registry access permission denied. Here is the stack trace:

```
 at ASCOM.Utilities.RegistryAccess.OpenSubKey(RegistryKey ParentKey, String SubKeyName, Boolean Writeable, RegWow64Options Options)\r\n   at ASCOM.Utilities.RegistryAccess.NewCode(Boolean p_IgnoreChecks)
```

And the accompanying message is:

```
OpenSubKey: Exception encountered opening key - Result: 0x5
```

I can probably build from source but I thought it important to point out that the release is crashing for me.

ClutchplateDude, 2020-08-03:
Do you have the latest ASCOM installed (I think 6.2) ?

BenJuan26, 2020-08-03:
That was the issue. I was getting my computers confused and didn't have ASCOM installed on this one. Still, catching the exception would probably be a good idea so that other people know what the problem is.

closed 2020-08-03 by BenJuan26

ClutchplateDude, 2020-08-03:
I think this might be too early to catch an exception. The program won't even start up if the dependencies are missing. Did you have to see any OATControl code in the stack when you looked at that exception?

BenJuan26, 2020-08-03:
> I think this might be too early to catch an exception.

I think you're right. There was very little info available, and I didn't see an OATcontrol file or line number mentioned anywhere. It complained about one of the constructors throwing an exception, but I don't recall whether it was OATcontrol-specific or some internal .NET class. I don't have the report handy anymore, and now that I've installed ASCOM I won't be able to reproduce it.

ClutchplateDude, 2020-08-03:
My problem also :-)