OpenAstroTech/OpenAstroTracker #68: Wrong Serial interface for DEC?

stone17, 2020-08-30:
https://github.com/OpenAstroTech/OpenAstroTracker/blob/d383d151e34142fe88fd5ae8a6d664c53c0af1c6/Software/Arduino%20code/OpenAstroTracker/Configuration_pins.hpp#L37

This should be Serial2?

My DEC motor doesnt work properly if this is set to Serial3.

pay4nothing, 2020-10-23:
Hi @Stone17,

thank you for your comment. I just had a weird behavor of the DEC axis with slipping and non holding the position. I fixed this to Serial2 and my problem was solved. So thank you!

andre-stefanov, 2020-10-23:
@ClutchplateDude this really looks like a type. I just wonder why it did not cause issues for e.g. @OpenAstroTech since the this change was done 3 months ago with [v1.8.02](https://github.com/OpenAstroTech/OpenAstroTracker/commit/308a0b3f82951cc9539fa4a7019a85a127707cfb)

pmneo, 2020-12-03:
RA and DEC can be both on Serial3. The have different adresses: 0b00 for RA (with MS0 = LOW and MS1 = LOW) and 0b01 for DEC (with MS0 HIGH and MS1 LOW) (https://wiki.openastrotech.com/wiki/NEMA_Wiring#TMC2209)

CS

andre-stefanov, 2020-12-13:
The issue was fixed by adapting the comment. Be aware that the most recent version of this file looks different because it is using separate pin number definitions now (to also allow SoftwareSerial).

closed 2020-12-13 by andre-stefanov