OpenAstroTech/OpenAstroTracker #138: Problem with printed parts

garnek1605, 2020-12-12:
Hello, 
first of all - thanks for that great project !
I'm still printing parts and found 2 issues 
1. \OpenAstroTracker-master\CAD\STEP\03_polemount_6801_v2.step - this part has 0,5 mm bottom instad of around 5 mm. Easy to fix just need to extrude the bottom to correct hight.  
[image]

2.\OpenAstroTracker-master\CAD\STEP\11a_splitring_part1_v4.step 
\OpenAstroTracker-master\CAD\STEP\11b_splitring_part2_v4.step 
\OpenAstroTracker-master\CAD\STEP\11b_splitring_part2_timingbelt.step 
This is more complicated. 2 Holes are missaligned (on the outer parts) and there is a problem with the perimeter around the joint. A picture is worth 1000 words. 

[image]
[image]

intercipere, 2020-12-12:
Hey, 

for no. 1 i assume you mean the side marked with "2" in your picture? that is supposed to be shaped like that, arguably a bit thin but its correct. 

For no. 2 you have the wrong middle part loaded, this one is the correct one: https://github.com/OpenAstroTech/OpenAstroTracker/blob/master/CAD/STEP/11b_splitring_part2_v4.step
(i know, i know, theres a file cleanup coming up :)

garnek1605, 2020-12-12:
Part 1, the problem I had was the bed adhesion, this 0,5 mm zones got ripped apart or deformed after I broke it off the bed. The "old" version of that part was 20mm thick. My idea was to extrude the new part thicker so i can break it off the bed in one piece. Is there going to be a problem of colliding with something i know nothing about yet ?

Part 2 solved, it will remain a mystery for me, how I might have missed it.Thanks for the help.
+

closed 2020-12-12 by garnek1605