OpenAstroTech/OpenAstroTracker #148: Wikifactory open hardware community welcomes OpenAstroTracker!

sandina, 2021-06-04:
Hey OpenAstroTech team!

Just recently, a community member of Wikifactory recommended your Project to join our collaboration platform for open hardware - with a community of over 100,000 designers, engineers and hardware innovators working on over 5K hardware and product innovation projects. 

I am co-founder of Wikifactory myself and agree that the greater community on Wikifactory and especially our space and astronomy enthusiasts, would not only learn a lot from your project but would like the chance of contributing and building upon your amazing work. 

As a social platform for collaborative product development, we definitely were inspired tools like Github and we focused on building a collaboration infrastructure built for product design from the ground up. 

A few months ago, we launched an Import from Github feature to make it very easy to bring projects that already exist on Github into Wikifactory in just a few minutes. 

I went ahead and imported your project to see how it would look, and whilst it is not listed for the broader community - I have the chance to show you how it would result! Take a look! https://wikifactory.com/@botler/the-openastrotracker

I would love your feedback. My hopes is that we could transfer the project to you so that it can be listed on our Platform for our community to find and engage with. 

We have built Wikifactory for projects like yours, so I would definitely care to know what you think and I would even suggest that we show you around in a Demo too!

At the heart of this message is that I would hope to be able to see your project evolve and grow with our community.

cameronswinoga, 2021-06-05:
@sandina Thanks for reaching out, wififactory looks interesting. We've had an internal discussion and decided that our current CAD workflow works well enough for our needs, but we will keep wikifactory in mind as a future platform option. :)

closed 2021-06-05 by cameronswinoga

sandina, 2021-06-06:
Thanks @cameronswinoga, I can totally understand, achieving a workflow like you have done is a feat and I can understand that a shift could compromise that. Really do hope to see you guys build something on Wikifactory one day, do let me know and I would be happy to give you guys a show around. I like to follow teams like yours, because it is what inspires us in building Wikifactory. 

We ultimately want to understand what it takes for real-time CAD collaboration online and how that can enable open source innovation in hardware and physical products. So I thought to share with you a glimpse of what i mean for future reference - https://wikifactory.com/@botler/the-openastrotracker/file/CAD/STEP/20deg/04_rollmount%20right_20%C2%B0_v5_2020base.step

Would you want me to take the project down from Wikifactory? It currently is under a Public Domain Profile we call Botler, attributing back to this repository and recognising you as authors. I can understand that you might not want to maintain the project on Wikifactory, but just wanted to triple check whether you would also like to remove the forked project from the Wikifactory platform. 

Look forward to seeing your progress, and thanks!

cameronswinoga, 2021-06-06:
@sandina yes please remove the project, thanks for checking with us :)