OpenAstroTech/OpenAstroTracker #152: When printing the threaded insert parts (TI), why are there still a bunch of non-TI holes?

JJ-White, 2022-02-09:
As you might gather from my previous issue I've been printing the parts for the base, and I've used the OAT Part chooser v1.5.1 to select the parts to be printed. I've chosen the option for threaded inserts, but after printing the base I notice only the holes that mount the 14_RAmotor_mount_NEMA_down bracket on the left and right rollmounts have the right dimensions for the inserts. The 4 holes to mount the rollmounts to the base all have non-TI dimension, and so does the hole on the bottom of the 02_angle_2020base part, which doesn't even have a TI version. 

I haven't printed any of the moving assembly yet, but I'm wondering why not all places where bolts thread into plastic have been changed?

crashmaxx, 2022-02-10:
I agree with using inserts everywhere possible.

I'd be willing to make the files if someone can help put together a list of parts that need to have TI versions made.

JJ-White, 2022-02-10:
So far the parts I found with missing TI changes are:
- 04_rollmount <left/right>_<degrees>_v6_2020base_TI.STL - These still have normal sized holes on the bottom of the parts, but are the correct size for the RA motor bracket mount.
- 02_angle_<degrees>deg_v4_2020base.STL - This part needs a TI version where the size of the hole in the bottom is increased.

I'm still in the process of printing the parts not in the base, but so far they all seem to have a correct TI version. I assume only the screws hidden on the bottom were overlooked when making the TI parts.

JJ-White, 2022-02-11:
Wouldn't it make more sense to also make them threaded? If you're printing the TI parts you will already have threaded inserts at hand, and IMO they make the build feel a lot higher quality. But I agree it's definitely not a big deal.

DRA6N, 2022-02-12:
> Wouldn't it make more sense to also make them threaded? If you're printing the TI parts you will already have threaded inserts at hand, and IMO they make the build feel a lot higher quality. But I agree it's definitely not a big deal.

You can slowly enlarge the holes with a drill bit and insert your own TI's.