OpenAstroTech/OpenAstroTracker #161: What version of Solidworks was used ?

demondev, 2024-01-03:
I'm looking at tweaking some of the parts to better suit my needs and I wonder what version of Solidworks was used to create the part files? Both year and license type (EDU, Commercial, Maker, etc.).

Tiring of the endless license changes Dassault (and the competition) like to make and the uncertianty that it leads to for hobbyists I also wonder if anyone has considered recreating the parts in a more open application ?

Awesome project by the way.

julianneswinoga, 2024-01-03:
I'm not the main mechanical designer (correct me if I'm wrong @intercipere :slightly_smiling_face: ) but I believe:
- the main OpenAstroTracker repo was designed with SolidWorks 2020 EDU
- Newer designs (OpenAstroExplorer) was/is designed in SolidWorks 2022 (not sure of license)

However all parts should be available as a STEP file for editing so you don't need SolidWorks. We've been kind of passively looking at alternatives but no serious work towards it.

demondev, 2024-01-03:
The problem with STEP files is the design tree is lost along with all the sketches etc. 

STEP files are ok if you need to produce a part or make small modifications but if you want to change anything patterned for instance then it's a pain.

I'd be interested to get contirmation of the versions used from @intercipere. I'm considering options to edit the parts.

intercipere, 2024-01-05:
Openastrotracker repo should still be 20-21 Edu. Some of the newer repos are 23-24 Maker license

demondev, 2024-01-06:
Many thanks for the info. I have a love hate replationship with Solidworks, the application is awesome the licensing (and file versioning) is very much not awesome. I'm very tempted to redraw the files in FreeCAD but I don't have the time. As you can tell I am somewhat conflicted at the minute :)