OpenAstroTech/OpenAstroTracker #162: FileNotFoundError: [Errno 2] No such file or directory

lynx2948, 2024-10-19:
[oat_fw_gui_2024-10-19-17-29-15.log](https://github.com/user-attachments/files/17447588/oat_fw_gui_2024-10-19-17-29-15.log)

The following Error message is diaplayed:
2024-10-19 17:37:45,098:CRITICAL:Exception caught 
2024-10-19 17:37:45,098:CRITICAL:Traceback (most recent call last):
  File "shutil.py", line 815, in move
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'D:\\Astronomy\\OpenAstro Tracker\\OATFWGUI_1.0.10-release+df7b94_Windows\\OATFWGUI_1.0.10-release+df7b94_Windows\\OpenAstroTech-OpenAstroTracker-Firmware-400a3cc' -> 'D:\\Astronomy\\OpenAstro Tracker\\OATFWGUI_1.0.10-release+df7b94_Windows\\OATFWGUI_1.0.10-release+df7b94_Windows\\OATFW'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\Astronomy\OpenAstro Tracker\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI\qt_extensions.py", line 39, in run
    raise e
  File "D:\Astronomy\OpenAstro Tracker\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI\qt_extensions.py", line 34, in run
    result = self.fn(*self.args, **self.kwargs)
  File "D:\Astronomy\OpenAstro Tracker\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI\gui_logic.py", line 200, in download_and_extract_fw
    self.logic_state.fw_dir = extract_fw(zipfile_name)
  File "D:\Astronomy\OpenAstro Tracker\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI\gui_logic.py", line 88, in extract_fw
    shutil.move(extracted_dir, fw_dir)
  File "shutil.py", line 835, in move
  File "shutil.py", line 434, in copy2
  File "shutil.py", line 254, in copyfile
FileNotFoundError: [Errno 2] No such file or directory: 'D:\\Astronomy\\OpenAstro Tracker\\OATFWGUI_1.0.10-release+df7b94_Windows\\OATFWGUI_1.0.10-release+df7b94_Windows\\OpenAstroTech-OpenAstroTracker-Firmware-400a3cc' 
2024-10-19 17:37:45,098:CRITICAL:

This happen regardless of the firmware selected for down load. 

Machine is running windows:
Edition	Windows 11 Home
Version	23H2
Installed on	‎10/‎1/‎2022
OS build	22631.4317
Experience	Windows Feature Experience Pack 1000.22700.1041.0

julianneswinoga, 2024-10-19:
Can you try moving the folder? There's a warning in the log:
```
2024-10-19 17:29:15,900:WARNING:Path D:\Astronomy\OpenAstro Tracker\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI_1.0.10-release+df7b94_Windows might
have too many characters in it (108)! Downloading/building firmware may create files with path
lengths greater than the default Windows path length of 260 characters.
If you get 'file not found' errors the easiest solution is to move
the D:\Astronomy\OpenAstro Tracker\OATFWGUI_1.0.10-release+df7b94_Windows\OATFWGUI_1.0.10-release+df7b94_Windows folder somewhere with less characters in the path.
```

lynx2948, 2024-10-20:
Restarted the OATFWGUI, loaded local config file, Selected 1.13.9 firmware and clicked download. Download completed successfully. selected MKS Gen L v1 clicked build FW. Build Firmware Completed successfully.

closed 2024-10-20 by lynx2948